SasaKaranovic/DIY-Camera-Slider #2: Wrong part list

Eisbaeeer, 2022-07-10:
Hi Sascha
I found a small mistake in part list. The slider only need´s one pulley, right?
https://github.com/SasaKaranovic/DIY-Camera-Slider/tree/master/Hardware#part-list

ghtomcat, 2022-07-16:
Also, it would help to link where to buy the endswitch

Eisbaeeer, 2022-07-16:
> And which bore the idler has (3mm or 5mm).
The idler in project has 3mm. I used standard 5mm and drilled up from 3 to 5.
You have to set the right hight with spacer.
Endswitches are the small one from e.g. Omron like D2FC.
Hope that helped!

ghtomcat, 2022-07-18:
@Eisbaeeer thanks for your help! I printed all the 3D Parts and now i'm waiting for the other parts to arrive

SasaKaranovic, 2022-07-23:
Thanks @Eisbaeeer !
The 280mm was used in the "old" version that uses 2040 aluminum extrusion instead of 2020. Good catch, changed it to 200mm.
And you are right, it's just one GT2 pulley.

@ghtomcat I believe the pulley I had laying around was 5mm bore and since I'm using 3mm screw I had to 3D print a tiny "sleeve" to keep pulley centered. If 3mm screw fits trough the 3mm pulley then I would say go with that.
Good catch! I have updated the BoM to include end-switch. Also I would love to see how your camera slider turns out!

thijstriemstra, 2023-04-12:
> Good catch, changed it to 200mm.

perhaps this ticket can be closed then.

closed 2023-04-14 by Eisbaeeer