SasaKaranovic/DIY-Camera-Slider #4: Pulley issue

ghtomcat, 2022-08-05:
Do i have the wrong pulleys or the wrong extrusion? Somehow the two do not fit :( 

[image]

[image]

[image]

buttim, 2022-08-23:
for my experience you might need to loosen the screws that fix the pulleys a bit. It might depends on your printer's tolerance too

ghtomcat, 2022-08-24:
Thank you @buttim, i will try that and report back